MosaicManufacturing/canvas-issues #4: Upload Printer Profile not working in Firefox

brandonbloch, 2018-08-21:
**Describe the bug**
The control to upload the profile is nonfunctional in Firefox. You can't get as far as even selecting the profile to load.

**To Reproduce**
Steps to reproduce the behavior:
1. Use Firefox
2. Attempt to upload a profile

**Expected behavior**
File selector should appear

**Operating System (please complete the following information):**
 - Browser: Firefox

brandonbloch, 2018-08-21:
See https://support.mosaicmanufacturing.com/hc/en-us/community/posts/360015295253-Upload-Printer-Profile-not-working-in-Firefox?flash_digest=71c4d2f2584d15506311549acc027ae7bd25bc91 for original report

ShadMan5, 2018-08-22:
It also did not work when accessing through MS Edge.

leemun1, 2018-08-22:
Thank you @brandonbloch @ShadMan5 , this issue has been acknowledged and fixed. The changes will be reflected on the next deployment of CANVAS.

closed 2018-08-22 by leemun1