MosaicManufacturing/canvas-issues #10: Minor Simplify3D Import Issues

gtjoseph, 2018-08-22:
**Describe the bug**
When importing an FFF profile that has Retraction disabled, the settings still get applied to the style and there's no corresponding checkbox to disable retraction.  you have to manually zero them out.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Simplify3D'
2. Create an FFF profile with retraction settings.
3. Uncheck the Retraction checkbox
4. Save the profile and export it.
5. In Canvas, create a new project
6.  Import the FFF just saved.

**Expected behavior**
Either there should be a similar checkbox in the style that is unchecked OR the retraction parameters should all be 0.

**Screenshots**
[image]

**Operating System (please complete the following information):**
 - OS: Fedora 28
 - Browser vivaldi
 - Version 16

**Additional context**
Also, a Z-Lift of 0.5mm was added to the retraction settings that wasn't in the Simplify3D profile at all.

Otherwise GREAT WORK!!!!!

brandonbloch, 2018-08-30:
Hi @gtjoseph, thank you for your kind words and for reporting this issue!

We were unable to replicate this behaviour—the code for profile import/conversion uses the logic you describe, where if retraction is unchecked the retract distance and z-lift should both be 0. However, we fixed a few bugs last week that were related to profile import in general. Would you mind trying this again and seeing if you're still having the problem? Please be sure to give the page a refresh to ensure you're running the latest version.

Thanks,
Brandon

brandonbloch, 2018-09-24:
Closing this issue due to inactivity. Please feel to re-open if necessary.

closed 2018-09-24 by brandonbloch