MosaicManufacturing/canvas-issues #11: Add a "Coast" feature.

gtjoseph, 2018-08-22:
**Is your feature request related to a problem? Please describe.**
The filaments I print tend to expand in the hotend and don't like to be retracted.  The result are blobs or stringing when the hotend does non-printing moves.

**Describe the solution you'd like**
A Coast feature that would stop extrusion a configuable number of mm instead of doing a retraction.

**Describe alternatives you've considered**

**Additional context**
Similar to Simplify3D's Coast feature.

gtjoseph, 2018-08-23:
The problem with wipe is that it's continuing to move over material already extruded for that layer so the material already in the nozzle keeps expanding out and just blobs up.  Let me experiment with more small retraction amounts.  Too much and the material will jam in the hotend.

brandonbloch, 2018-08-27:
Thank you @gtjoseph, I see the difference in what you're talking about. We will look into adding this functionality!

brandonbloch, 2019-06-26:
The ability to set a coasting distance for perimeters has landed in CANVAS! You can find the setting in the General tab under Perimeter Seams. As such I'm closing this issue, but feel free to re-open it if you find that you still need the ability to coast on non-perimeter paths after trying it out.

closed 2019-06-26 by brandonbloch