MosaicManufacturing/canvas-issues #7: Side transitions still build transition tower

Kanshisha, 2018-08-22:
**Describe the bug**
Using Canvas (beta) when side-in-place transitions are selected for printer profile and for project a transition tower is still created in the main print area ( in this case the print area is very small and side transitions are the only option.

**To Reproduce**
Steps to reproduce the behavior:
1. create new printer profile with side transitions set
2. create new project and select side transitions
3. create model with multi materials
4. view transition tower even though you do not want one (nor does it actually fit on the build plate, instead it extends it past the edges of the build plate)

**Expected behavior**
Side transitions create erm.... side transitions?
also, please note that there is no current ability to specify the extrusion rate for side transitions.

brandonbloch, 2018-08-22:
Hi @Kanshisha, thank you for the bug report! We will look into this ASAP.

As for the extrusion rate for side transitions, this setting is actually found within style profiles rather than the main printer profile. The reason for this is because the speed may be dependent on other print settings like temperature, while settings like the physical purge location would stay the same. You can find that setting in the Transition section, under Side Transition Settings in the bottom-right.

brandonbloch, 2018-09-24:
This issue has been resolved in our most recent update. Feel free to open another issue if you're still having problems with it, but the use of side transitions should now be accurately reflected.

closed 2018-09-24 by brandonbloch