MosaicManufacturing/canvas-issues #9: Arrow keys

cguerin, 2018-08-22:
**Is your feature request related to a problem? Please describe.**
Using buttons and typing in X, Y, Z sucks.

**Describe the solution you'd like**
Use arrow keys on keyboard to gently move objects and rotate objects using the appropriate tools

**Describe alternatives you've considered**
Typing portions of numbers and backspacing  19 + <bs> + 18 + <bs> + 17, etc... annoying!

Thanks guys, keep up the good work!

brandonbloch, 2018-08-23:
Hi @cguerin, thank you for your submission!

I agree that this would be very nice. Are you looking for this arrow key functionality while in the X/Y/Z text boxes, or just whenever a model is highlighted?

cguerin, 2018-08-23:
Great! Whenever a model is highlighted, perhaps shift + cursor could affect the z axis?

leemun1, 2018-08-27:
Hi @cguerin , thank you for your submission! 
As a response to your request, we implemented a feature to transform (translate, rotate, scale) objects via pressing the ARROW-UP/DOWN keys. Once you open up a tool panel and click into the input field for an axis, the axis will be then "selected" and you will be able to make adjustments on the axis with your keyboard. Also note that hitting the ARROW-UP/DOWN keys while holding down the SHIFT key will make adjustments of larger sizes. 

This feature will soon become available to you in the next deployment of CANVAS. Thank you!

closed 2018-08-27 by leemun1