MosaicManufacturing/canvas-issues #12: Independent Printer, Material and Support styles

rcsun, 2018-08-23:
**Is your feature request related to a problem? Please describe.**
Having to set multiple styles for the same printer based on different material or support usage is tedious.

**Describe the solution you'd like**
Styles that are related to printer settings, extrusion styles, material properties and support should be set independently of each other, then for each model, drop down menus to select what style should be used in each category.

brandonbloch, 2018-08-30:
Hi @rcsun, thank you for the request!

We are working on a material profile system that will allow overriding certain print settings, such as temperature, retraction, extrusion multiplier, and so on.

There is currently no plan for profiles that specifically handle support settings, but this sounds interesting. What kinds of settings would you like to see here? Would it basically be all the settings found in the Supports section of style profiles? Or is there more/less that you would like to see?

Thanks,
Brandon

rcsun, 2018-08-30:
Maybe my initial post wasn't very clear, but to elaborate: 

The way I see setting up a print whether it be multi color/ material or even single color/material would consist of starting a new project. Selecting my printer of choice. 

Then from there, instead of selecting a single FULL settings profile every time. I'd like to have finer granular control in selecting from categories.

Imagine this: You go to Chipotle, instead of picking a type of wrap/container (burrito or bowl) then picking your meat (beef, chicken pork), then picking your toppings; you are presented with a massive menu preset with combinations of everything they have to offer (i.e. chicken burrito with xyz topings, chicken bowl with xyz toppings, beef burrito with xyz topings, beef bowl with xyz toppings). From there you can modify little things. The list would be massive.

Now that's not very efficient, is it? That's essentially what we are doing here. Instead, categories should be clearly decoupled from each other in the styles selection menu. Choose from presets for each category of style.

Hopefully this makes more sense?

As for supports, it would be nice to manually add or remove supports.

brandonbloch, 2018-09-05:
Hi @rcsun,

You make a very good point. We will certainly keep this in mind as we improve and refine the platform!

As for manually adding/removing supports, this feature is planned further down the road. Please feel free to make a separate issue for that feature request if you would like to keep track of its status.

Cheers,
Brandon

brandonbloch, 2018-10-24:
Hello,

I'm updating this issue as we have shipped full material profile support. You can find its UI in CANVAS on the Color page by clicking on a color swatch.

At this time we do not plan to incorporate style settings for supports. However, this may change as CANVAS matures. If you would still like to see this, please feel free to open a new issue to keep it on our radar (if you do, be sure to provide a [link to this issue](https://help.github.com/articles/autolinked-references-and-urls/#issues-and-pull-requests)).

closed 2018-10-24 by brandonbloch