MosaicManufacturing/canvas-issues #13: Add print size (x, y, z coordinates)

binux14, 2018-08-24:
**Is your feature request related to a problem? Please describe.**
Not a problem but I am not sure how big my print will actually be.

**Describe the solution you'd like**
It would be nice to have a the size of the print on each of the 3 axis (x, y, z). 

**Describe alternatives you've considered**
No alternatives considered.

**Additional context**
N/A

brandonbloch, 2018-08-30:
Hi @binux14 and @biscuitlad,

Thank you for your feedback! We will be adding this functionality in our next update. You will be able to see the combined bounding box of whatever is selected in the Place step, as well as the overall bounding box of the print in the Preview step.

In the meantime @biscuitlad, I would recommend using http://gcode.ws to find out the bounding box of the print (under the "Model info" tab once you load a file). I hope this works for you until we build the functionality directly into CANVAS.

Cheers,
Brandon

brandonbloch, 2018-09-11:
Hi @binux14 and @biscuitlad,

We have just put out an update to CANVAS that allows you to scale a model to certain dimensions (in mm) in addition to the scale factor. The dimensions will update based on what models are selected, so by selecting all models you can see the overall build dimensions of the print.

[image]

We'll keep this issue open until we have build dimensions included in the print summary. In the meantime, hopefully this is sufficient for your use cases!

Cheers,
Brandon

biscuitlad, 2018-09-11:
Cool, many thanks. I'll get testing!

closed 2018-09-23 by brandonbloch