MosaicManufacturing/canvas-issues #14: Things seem to be broken today

gtjoseph, 2018-08-24:
**Describe the bug**
I can create a new project and add an stl file but can't move it.  Canvas was stuck on then finally threw a javascript error saying there was a problem with move and to reload the page.  After a reload, I can do nothing to the project.  Can't clock on the actions on the left and I can't even add another stl file.

Cleared cache, tried Chrome instead of Vivaldi, same issue.

The JS console has these messages...
```
service-worker.js:1 Couldn't serve response for "https://canvas3d.io/" from cache: Error: The cached response that was expected is missing.
    at https://canvas3d.io/service-worker.js:1:4662
(anonymous) @ service-worker.js:1
Promise.catch (async)
(anonymous) @ service-worker.js:1
The FetchEvent for "https://canvas3d.io/" resulted in a network error response: the promise was rejected.
Promise.catch (async)
(anonymous) @ service-worker.js:1
service-worker.js:1 Uncaught (in promise) TypeError: Failed to execute 'fetch' on 'ServiceWorkerGlobalScope': 'only-if-cached' can be set only with 'same-origin' mode
    at service-worker.js:1
```

gtjoseph, 2018-08-24:
Ha!  Yeah it's working now.  I must have hit it at just the right (wrong) time. :)

george

brandonbloch, 2018-08-24:
Our apologies, but I'm glad to hear it's working now!

closed 2018-08-24 by brandonbloch