MosaicManufacturing/canvas-issues #15: Just to confirm - Single material splicing currently?

awr4sc, 2018-08-25:
I do not see any options to splice different materials (tpu, pla, etc) or settings like in Chroma; i'll assume this is a future development?

Also the slider bars on the bottom of the screen and right of the screen do not seem to do anything once you progress to the preview window. Something to preview layers at a later point?

[image]

awr4sc, 2018-08-26:
Just a bit of testing and a few more questions? or requests.  Print summary tells you the drive, and length, but not the colour; could you identify the colour for the drive in the summary. 
[image]

Also there does not appear to be any way to label the colours, so, when loaded into palette+ the are listed as "default".

brandonbloch, 2018-08-30:
Hi @awr4sc, you're correct on both counts.

Material profiles (along with splice settings) are still in development, but rest assured that they are high on our list of priorities. This will also include the ability to name the four inputs, which would show up on Palette.

The sliders are indeed for previewing the toolpaths created when slicing (the intent is that the Preview step visualizes these toolpaths instead of the solid models); this preview is currently in development as well.

Good idea re. showing the color for each drive! We will add that to our list.

As for the drives appearing backwards, which slicer were you using for the Chroma print? If the slicer lets you choose the actual tool for printing each model, then this would be suspicious, but if you simply tell it which models to print in the same color, it likely decides which tool to use for each group in its own way, which could depend on a number of factors based on how the logic is implemented. **TL;DR** it shouldn't be a problem, but that is interesting to note.

Cheers,
Brandon

closed 2018-08-30 by brandonbloch