MosaicManufacturing/canvas-issues #19: Support interface material

lowfuel, 2018-09-01:
**Describe the solution you'd like**
To help reduce the amount of expensive soluble material, generate supports where only the last few millimeters are of a different material.

**Describe alternatives you've considered**
Not really an alternative, but it's a feature they've added to Slic3r Prusa Edition and I think it would be great for other printers using Palette 2.

**Additional context**
See this article: https://www.prusaprinters.org/printing-soluble-interface-supports-prusa-i3-mk2-multi-material/

brandonbloch, 2018-09-05:
Hi @lowfuel,

Thanks for suggesting this—we really want to see this too!

We are actually working on material profiles as we speak. When we push out the update that introduces them, CANVAS will fully support soluble supports and/or support interfaces.

Cheers,
Brandon

lowfuel, 2018-09-05:
Thanks fantastic, Brandon! Thanks!

brandonbloch, 2018-10-24:
Hi @lowfuel,

We shipped material profile support earlier this week, and have finished the deployment off today with the inclusion of options for setting the extruder used for the raft, supports, and support interface (independently). You can find the former under First Layer and the latter two under Supports.

If you're using soluble supports, the best way to accomplish this would be to reserve one of the four inputs for your soluble filament and use that input as your support and/or support interface extruder but not use it for any of your models.

As this feature has shipped, I'm closing this issue. If you experience any problems or if anything with the workflow is off, feel free to open a new bug report and link to this issue.

closed 2018-10-24 by brandonbloch

lowfuel, 2018-10-24:
Thanks, that's awesome!