MosaicManufacturing/canvas-issues #24: Splice Weld Factor

trevorray51, 2018-09-18:
The feature I am requesting is to have the ability to modify the the heat and compression factor for individual materials. On my Palette+ I have a Heat and compression factors for PLA set as 3 and 4 respectively. I went to print a model using Canvas3D this weekend and the print failed due to a weld breaking. I went through Canvas3D and could not find anywhere in the settings where material profiles could be set up for different parameters.

brandonbloch, 2018-09-18:
Hi @trevorray51,

I'm sorry about the confusion and the failed print. We are still actively developing the material profile system of CANVAS, and along with overrides for slicer settings, these profiles will include splice settings for Palette printing. Expect to see material profiles appear in the coming weeks! For now, however, custom splice settings with Palette will still require using Chroma.

trevorray51, 2018-09-18:
Thanks for the quick response. Good to hear. Ya I used canvas instead of chroma because chroma kept failing to make the files saying it ran out of memory. It was odd because the file only had color change at certain layers.

closed 2018-09-18 by trevorray51