MosaicManufacturing/canvas-issues #25: HTC Vive starts on new project

DeltaFox1980, 2018-09-22:
**Describe the bug**
Every time the page is loaded and a new project is started (currently no projects) the page makes my HTC Vive turn on. - Addition: if Canvas creates a 'new tab' (in the same window, i guess, since i couldn't see it.) the Vive is started.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://www.canvas3d.io/
2. login
3. create a new project
4. select a printer
5. select a style
6. the Vive starts
--
1. enter Canvas and a project
2. select something that will create a tab
3. the Vive starts

**Expected behavior**
- the Vive shouldn't start. 
- Since it's not needed, unless you plan to allow Vive users to modify/edit/transform the models in 3D (Cool Idea, but i don't think you want to do that, but it should happen after the user selects and uploads the model and accepts (first time with option of always starting VR after upload or only this time after upload or never, to view and edit the model in VR ).

**Screenshots**
not really possible...

**Operating System (please complete the following information):**
OS: Windows 10
Version: Latest
-
Browser: Firefox
Version of browser: 62

**Additional context**
Own HTC Vive. ?

brandonbloch, 2018-09-24:
Hi @DeltaFox1980,

Thank you for the report. Unfortunately since we don't have a Vive this will be a hard issue for us to reproduce and try to fix, but we'll see what we can do.

Can you test to see if this happens with other browsers, or just with Firefox? That should help us get started.

Thanks,
Brandon

DeltaFox1980, 2018-10-04:
> HI, Sorry for the delay, my work was demanding. - The issue is only with
> firefox. Edge and Internet Explorer didn't have any issues with starting VR
> (but other problems). So it's only Firefox (I don't have chrome).
>
> I heard a mention of EU based people (from the live stream), are any in
> Germany that could/can help with the problem? - if someone can help, i'd be
> willing to lend my Vive for a few days for testing and maybe a few new
> ideas for 3D printing...
>
> Thanks for the work!
> Benny

for info: edge and internet explorer are both not functioning. - Loading 2
files for multi-color printing fails (for Firefox, edge and IE).

DeltaFox1980, 2018-10-04:
I've tried to apply a custom material/color to anything, but it fails, or
doesn't register. (tested with: Edge, Firefox, Internet Explorer.)

brandonbloch, 2018-10-05:
Hi @DeltaFox1980,

That is a new issue for us. Can you send a link to your project to canvas@mosaicmfg.com so we can see what's going on when you try and apply colors/materials?

Since this is a separate issue, if you'd like to keep track of it here on GitHub please open a separate bug report dedicated to that issue.

brandonbloch, 2018-10-05:
Unfortunately our development team is entirely located in North America, but thank you so much for offering to lend your device! Since this issue seems to be related to WebVR (which Firefox has the most support for, possibly explaining why the 3D renderer triggers it to enter VR mode) we will try and put together a device we can use for testing. Long-term, a VR UI for CANVAS would be very cool!

As for new printing ideas, we're always happy to discuss possible new features and printing techniques! Feel free to open feature requests for ideas you may have, so that others can come along, discuss, and add to them. You can also always get in touch at canvas@mosaicmfg.com

Thanks so much!
Brandon

DeltaFox1980, 2018-10-06:
Thank you very much for the answer!

If you had the chance to test a VIVE, would you like to? To see what's
possible?  Etc.?

I'm asking because I'd be willing to send my VIVE to you for testing and
making new options available for VIVE/Rift users.

brandonbloch, 2018-10-30:
Hi @DeltaFox1980,

Thank you for the offer. Based on our roadmap of planned features we may not get to this for a fair while, but we will keep this in mind when we do get a chance to investigate a proper VR interface for CANVAS!

As for the original issue that kicked this thread off, can you please try this out (taken from the [Mozilla forums](https://support.mozilla.org/en-US/questions/1174025)):

- Type in the address bar `about:config` and hit Enter
- Promise to be careful, if asked
- Look for the preference `dom.vr.openvr.enabled`. Its value should be set to `false`, which is the default

Hopefully this prevents Firefox from sporadically launching your headset automatically, while still allowing you to launch it manually. Let us know if that works!

Cheers,
Brandon

DeltaFox1980, 2018-10-31:
Hi, the setting is true, so I'll set that to false. - it stopped starting
my vive. :) Thank you! - if/when you need a test subject for 3d - VR let me
know.

Again, thank you very much for everything!
Benny