MosaicManufacturing/canvas-issues #26: Cannot Save a project name

wwarwick, 2018-09-23:
**Describe the bug**
When trying to edit and save a project name, I am unable to.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Projects
2. Click on any project as though you are going to open it.
3. Click on the "Edit Project icon".
4. Edit the project name.
5. Click on save.

**Expected behavior**
The project name should update to the edited text but does not change from the original.

**Screenshots**
I enter the project name.
[image]

I click save and get the following result.
[image]

**Operating System (please complete the following information):**
 - OS: Windows 10 Professional
 - Browser: Google Chrome
 - Version: 69.0.3497.100 (Official Build) (64-bit)

**Additional context**
None

brandonbloch, 2018-09-24:
Hi @wwarwick,

Thank you for the report—we have confirmed the issue and will fix it in our next update (early this week). In the meantime, you can edit the project name from within the project:

[image]

Cheers,
Brandon

brandonbloch, 2018-09-24:
We have fixed this issue in our latest update. I'm closing this as fixed; if the problem persists, feel free to re-open it.

Thanks,
Brandon

closed 2018-09-24 by brandonbloch