MosaicManufacturing/canvas-issues #32: Transition tower located in the middle of the bed

HangarZero, 2018-10-03:
Hi all.
I'm having some issues with Canvas when generate the gcode.
The transition tower is allways in the middle of the bed….maybe I'm doing something wrong, but...I've tried several times and same result ever

[image]
Best regads

HangarZero, 2018-10-03:
...just find how to move the tower...but still an issue.
The "tool" to select the transition tower doesn't appears in google chrome in my laptop (17" screen).
Something related with resolution?. 

And...if it's not optimized for Chrome...see last pic...Microsoft explorer.

I still consider this as a bug...;)

[image]
[image]
[image]

Best regards.

brandonbloch, 2018-10-05:
Hi @HangarZero,

We are aware of the auto-positioning of the transition tower being incorrect and are working on adding smarter auto-positioning logic that is still faster than the full auto-arrangement. We'll update this issue as this work progresses.

I'd just like to confirm, does the transition tower preview appear in this location even though it overlaps with the print? The tower being non-representative of the final position would be a different issue altogether, one we have not seen.

We will be releasing an update today that reduces the vertical size of the model manipulation tools, which should prevent the auto-arrange button being too low. Please also note that you can use the Move tool to manually position the transition tower, as well as the Scale tool to adjust its size!

Lastly, CANVAS does not currently support Internet Explorer—we do fully support Chrome, Firefox, Safari, and Microsoft Edge, so I would recommend using Edge over IE when using CANVAS.

Cheers,
Brandon

brandonbloch, 2018-11-27:
This issue has been addressed and considered fixed as of yesterday. If there are still problems with the new approach, feel free to re-open this issue!

closed 2018-11-27 by brandonbloch