MosaicManufacturing/canvas-issues #39: Transition tower off bed when changing settings

binux14, 2018-10-29:
**Describe the bug**
When you open a multi-material project and go to Settings/Style Settings/Transition/Transition Method and you change it, when you go back to "Transition tower" method the tower on the visualizer moves out of the bed. I think the center of the towers moves to the origin.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a multi-material project
2. Click on the "Settings" tab
3. Click on "Transition"
4. Change the Transition Method to "Side transitioning" or "No handling"
5. Change the Transition Method back to "Transition tower"
6. The tower will now be off the bed.

**Expected behavior**
The tower should be in the same place as before changing the method.

**Screenshots**
[image]
[image]

**Operating System (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Version: 69.0.3497.100

brandonbloch, 2018-10-29:
Thanks @binux14, this is a known issue and we are actively working on a solution. The full auto-arrange solves this; however, this is slow and forces us to interrupt your coloring flow with a loading spinner. We want to come up with something that works as close to auto-arranging as possible without the flow interruption, to get the best of both worlds. We will update this thread as progress is made!

brandonbloch, 2018-11-21:
Closing as duplicate of #32

closed 2018-11-21 by brandonbloch