MosaicManufacturing/canvas-issues #40: Being able to modify e-steps for gpx processing

Bleky1, 2018-11-02:
**Describe the solution you'd like**
Since some of us have different extruders on our printers that use x3g format (Creator Pro with Bondtech for example) it would be great if we could modify the e-steps for the gpx processing like we can in slicers like S3D.

**Describe alternatives you've considered**
Workaround for this is manually processing the gcode file using GPXUI where you can add the correct e-steps.

HFMan, 2018-11-20:
Can't you put commands into the start sequence to do this?  How do you do it in S3D?

Bleky1, 2018-11-20:
Nope,S3D outputs the gcode and then calls the gpx plugin to process the gcode into binary code (just numbers).
If you have a printer that uses that format (Makerbot Sailfish FW) then e-steps config is under Firmware Configuration —> X3G

brandonbloch, 2018-11-21:
This is definitely important. As you may have guessed we use GPX after slicing to convert the generated GCode to X3G, similar to how both S3D and Chroma do it. In Chroma we enabled users to provide full GPX config files, allowing them to override any settings they wanted. However, this turned out to be fairly error-prone due to the different config file formats used by different forks of GPX. Moreover, we found that users only ever wanted to adjust their E-steps so this feature was fairly overkill. As such, we'll be taking a similar approach to S3D where they allow you to adjust just a few settings, so that users don't have to fumble with these complex files.

brandonbloch, 2018-11-30:
Hi all,

This functionality is now available in CANVAS. When the file type is set to `x3g`, you will be able to override the steps/mm values for both the A-axis (right extruder/T0) and B-axis (left extruder/T1):

[image]

Setting a value of `0` will simply use the value in the GPX profile you selected instead of overriding it.

closed 2018-11-30 by brandonbloch