MosaicManufacturing/canvas-issues #42: Unexpected error occurred while slicing

parkernbrown, 2018-11-20:
When using Canvas to slice a multicolor model, it hangs at 0% then jumps to 28% and processes until 100%. At that point it reports "An unexpected error occurred while slicing."

I'm not sure how to have you reproduce this, since it seems specific to my model. Originally it generated a sliced product but everything dropped to the build plate instead of staying aligned. After attempting to fix this, it now causes this error.

Screenshot:
[image]

Machine: HP Pavilion Gaming Laptop 15-CX0058WM
OS: Windows 10 Home x64, Version 1803, Build 17134.407
Browser: Google Chrome x64, Build 70.0.3538.102

brandonbloch, 2018-12-06:
Hi @parkernbrown,

Sorry to hear you've run into this issue. Can you confirm that this issue still occurs? If so, would you mind sending us a help package (right sidebar > Get Help with CANVAS) along with the STLs in your project?

parkernbrown, 2018-12-06:
Hello!

Thanks for getting back with me. I haven't seen the problem again, I ended
up deleting the project and starting a new one once my Palette got here
last night, so I'll definitely let you know if it occurs again.

Thanks,
Parker

brandonbloch, 2018-12-06:
Glad to hear everything's working! Apologies for the delay in getting to this issue.

I'll consider this issue resolved, but feel free to re-open it if you do encounter it again. And of course, thank you for letting us know!

Cheers,
Brandon

closed 2018-12-06 by brandonbloch