MosaicManufacturing/canvas-issues #44: Need help widgets or 'hover' help for most settings.

HFMan, 2018-11-20:
There are extrusion settings that I have no idea what they mean.  It seems to me that **_all_** settings should have some kind of 'hover over' help pop up to describe what it is.

The extrusion parameters I am referring to are:

1. Gap fill threshold
2. Retraction threshold

brandonbloch, 2018-11-21:
Hi @HFMan,

You're absolutely right—this is a planned improvement to CANVAS, but as it's not there at the moment let me at least address your two settings with some explanation:

**Gap Fill Threshold**

Gap fill refers to extrusion paths that are narrower than your extrusion width, which are generally used to fill in areas with small details. This setting allows you to specify the minimum XY length that the path must be for the slicer to include it in the final output. For instance, if your gap fill threshold is 2 mm, a path of 3 mm will be included but a path of only 1.5 mm will be dropped. A value of 0 for this setting means that all gap fill paths will be included, no matter how short they are. This seems ideal but the main reason to be able to raise this value is that gap fill paths are generally printed after other infill or solid paths, which means retraction and travel between them all, and the retractions with little extrusion between them can cause your extruder's drive gear to grind on your filament in the same spot and eventually lose traction.

**Retraction threshold**

This setting is similar to Gap Fill Threshold, and the two can be adjusted in parallel to overcome some of the limitations I mentioned above. Normally when printing a path, you will (1) move to the start, (2) un-retract filament, (3) print the path, (4), retract filament, and (5) move to the start of the next path. This setting allows you to specify the minimum amount of filament usage required to retract when finishing a path (and subsequently un-retract when starting the next path). This setting basically exists to directly combat the "drive gear grinding down your filament" issue. Using it can be a trade-off between quality (not retracting causes "strings" along the path of travel, as the filament in the nozzle is still slightly attached to the path your just printed) and reduction of filament grinding.

As you can see, these settings are powerful yet complicated, and we are working on a good way to explain not only what a style setting does to your print, but how and why you may want to adjust it, as well as how it interplays with other settings. We think these are just as important when teaching users how these various settings work. This is the main reason CANVAS doesn't have any tooltips around these settings yet—but I digress. I hope that helps clear things up, and if you have any other questions feel free to post again.

Cheers,
Brandon

brandonbloch, 2019-06-26:
As of yesterday, every printer and style setting in CANVAS has an explanatory tooltip on hover!

closed 2019-06-26 by brandonbloch