MosaicManufacturing/canvas-issues #46: Internal Server Error

steve220, 2018-11-22:
**Describe the bug**
When rotating a model, I get an "Internal Server Error"

**To Reproduce**
Steps to reproduce the behavior:
1. Place model
2. Click on Rotate
3. Try rotating the model
4. See error

**Expected behavior**
Model rotates without an Error

**Screenshots**
[image]

**Operating System (please complete the following information):**
 - OS: Win 7
 - Browser Chrome - Version 70.0.3538.110 (Official Build) (64-bit)

**Additional context**
Add any other context about the problem here.

brandonbloch, 2018-11-22:
Hi @steve220,

Thanks for reporting this! We discovered this issue today and will be rolling out the fix this afternoon. I’ll follow up when the rollout is complete.

Cheers,
Brandon

brandonbloch, 2018-11-22:
Following up to let you know this update has been rolled out. Thank you for your patience!

I'll close the issue but if you do still run into problems please feel free to re-open it.

Cheers,
Brandon

closed 2018-11-22 by brandonbloch