MosaicManufacturing/canvas-issues #49: Canvas reports Oops Something Went Wrong after profile import. Stuck now.

jamesarm97, 2018-12-04:
After starting a new Project, I tried to import a S3d profile and it showed the settings from the profile and I clicked Next and then it crashed and just shows "Oops, something went wrong" no matter what I do when I click on new Project. I tried to log out and back in and it still won't let me do anything.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to New Project
2. Click on Import Profile
3. Uploaded S3d fff export
4. Assigned Name and Icon then Next
5. Viewed all the imported settings then Next
6. Oops message displayed and cannot add a new project.

**Expected behavior**
It should work after importing profile or at least let me start a new one.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Operating System (please complete the following information):**
 - OS: OSX 12
 - Browser Safari

[image]

brandonbloch, 2018-12-06:
Hi @jamesarm97,

Sorry to hear you're running into this issue. Does refreshing the page at least get rid of the Oops?

Would you be willing to send your FFF profile to canvas@mosaicmfg.com so we can take a look at what's causing this error to occur?

Piebloxxer, 2019-04-03:
Although this is somewhat of an older ticket, I'd like to +1 this thread. 

I'm having a similar issue (also using Safari on macOS 10.14.3, image attached); however, I get the "Oops..!" immediately on login. All links on the page (and sidebars) are completely broken, save for the Logout and "Report a Bug" links (which is how I arrived on this repository in the first place.)

I would add steps to reproduce, but I unfortunately do not have any since I can only log into my account. 

I'm going to try logging into my account on other browsers/machines to see if the issue persists, but I thought it might be good to mention that the error exists in Safari. 

Thanks

[image]

brandonbloch, 2019-04-10:
Hi @Piebloxxer,

I believe this issue has to do with a bug in the onboarding tutorial. Can you please try clearing your browser's storage for canvas3d.io (Preferences > Privacy > Manage Website Data...) and see if reloading the page fixes the error?

Piebloxxer, 2019-04-10:
@brandonbloch Looks like it did.

Either way, it looks like the software works in Chrome; haven't had any problems with it as of yet.

closed 2019-06-26 by brandonbloch