MosaicManufacturing/canvas-issues #50: multiple print settings

nxt-dimension, 2018-12-05:
**Is your feature request related to a problem?
Currently there is no way to print multiple objects at the same time with different style settings.
Example: I want to print some text on top of my model. I want the text solid but the model with 20% infill.
Differentiating things like heat, speed, shells and infill can become critical to achieve perfect prints.

**Describe the solution you'd like**
Style settings should be connected directly with individual models instead of the entire print.

**Describe alternatives you've considered**
S3D allows this by splitting the models into individual processes. These processes can all have different settings and allow much more versatility to the over all model.

nxt-dimension, 2018-12-05:
I have just found the style override button for different colours/materials/ "Drives". This could be somewhat used to accommodate this request but not completely. 
Example: I want to print 5 cubes. 3 are, red, blue and green. I then want to print 2 purple cubes with different infill % each, I have already used 4 different drives and don't have room for more drive selections. I would still need to have alternative style settings "per object."
Style settings should be connected to objects instead of drives. This would add HUGE versatility to CANVAS.

mylife4aiurr, 2019-01-23:
I would also love this feature