MosaicManufacturing/canvas-issues #54: MKsum error mid printer - Printer Disconnected from OctoPi

furicks, 2018-12-08:
**Describe the bug**
A clear and concise description of what the bug is.

Error message from OctoPrint Plugin
"Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: WrN87692 Mksum"

**To Reproduce**
Steps to reproduce the behavior:
Printing from Octoprint V1.3.9
Sliced with Simplify3D V4.1.1
Chroma V3.1.1
3D Printer Electronics Arduino Due - RADDS V 1.5

**Additional context**
Gcode attached

[1-4in Angled Socket Hooks.mcf.gcode.txt](https://github.com/MosaicManufacturing/canvas-issues/files/2659920/1-4in.Angled.Socket.Hooks.mcf.gcode.txt)

brandonbloch, 2018-12-10:
Hi @furicks,

Does this problem happen reproducibly or was it a one-time error? I suspect that it may have been a communication-related issue rather than a GCode problem, but I may be mistaken. Please let us know and we can go from there.

furicks, 2018-12-10:
I have not seen it happen again, but I haven't tried reprinting exactly the same file either. I've never seen that error before and I've been using the same OctoPi setup for years. I would mark it a low priority, and I'll report if it happens again.

brandonbloch, 2018-12-10:
Sounds good! Thanks for the quick follow-up. I'll close this issue for now but please feel free to re-open it if you run into this again. I don't want to discount the possibility that it is a slicer problem.

closed 2018-12-10 by brandonbloch