MosaicManufacturing/canvas-issues #56: Longer transition tower

totalitarian, 2018-12-14:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I can't seem to completely bleed out black filament even at 230mm
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Ability to increase transition length up to 300mm
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

brandonbloch, 2018-12-14:
Hi @totalitarian,

Done! The maximum is now 330 mm.

I would exercise caution raising it all the way to 330, especially if you're using Palette 2 in accessory mode. The reason for this is that it becomes more difficult to generate reliable ping sequences on such large towers. This is not a problem if using connected mode (with a CANVAS Hub).

Cheers,
Brandon

closed 2018-12-14 by brandonbloch

totalitarian, 2018-12-14:
Thank you very much!

brandonbloch, 2018-12-14:
Yes, but there's not likely to be a Chroma update for at least the next few weeks. What you can do, however, is edit your printer profile's file manually:

1. Quit Chroma
2. Navigate to the profiles folder in Chroma's config directory
   - Mac: <home folder>/Library/Application Support/Chroma/profiles
   - Windows: <drive>\Users\<username>\AppData\Roaming\Chroma\profiles
3. Find your printer profile and open it in a text editor
4. Manually edit `purgeLength`, `minPurgeLength`, and/or `initialPurgeLength`
5. Re-launch Chroma and use as normal

Note that if you use Chroma to edit this printer profile, these settings may or may not get capped back to the app's maximum and you'd have to follow this process again.

totalitarian, 2018-12-14:
Thanks Brandon :)