MosaicManufacturing/canvas-issues #57: Time stamp in the future

jmsjgill, 2018-12-16:
Using CANVAS to slice multicolor prints for use in accessory mode produces files with a timestamp in the future.

**To Reproduce**
1. Start a project in CANVAS using the default CR-10 profile
2. Ensure the profile defines Palette use in accessory mode
3. Slice project
4. Click Download
5. Resulting files have time stamps in the future

**Expected behavior**
Expect file time stamps the correlate with either creation on the host (CANVAS) or with creation on the downloaded client

**Operating System (please complete the following information):**
 - OS: MacOS High Sierra (10.13.6)
 - Browser Safari 12.0.1

**Additional context**
None

jmsjgill, 2018-12-16:
[image]
Screenshot attached.

brandonbloch, 2019-01-07:
Hi @jmsjgill,

Thanks for reporting this! Unfortunately I'm unable to reproduce this issue—in all the various Palette modes, all my files have the correct timestamp. I'm on 10.14.2 Mojave and Safari 12.0.2.

What time zone do you live in? I'm in the same time zone as our servers (Eastern) so that may be the difference here.

jmsjgill, 2019-01-07:
I’m in MST (Utah). I didn’t think to check against the time of day but I’ll verify tonight.

closed 2019-06-26 by brandonbloch