MosaicManufacturing/canvas-issues #60: Palette 2 Plugin and Palette 2 Device Ping/Pong Report Mismatch

rsvp95, 2018-12-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Operating System (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

rsvp95, 2018-12-22:
Describe the bug
46% into the print Plug in reports "Latest Pong: Pong 1  100%" whereas Device reports 11 Pongs

To Reproduce
Run and Calibration print and watch the plugin and device statistics reports

Expected behavior
Statistics reports in plugin should match the device

Screenshots
Attached, please find.

Operating System (please complete the following information):

OS: Windows 10 Pro
Browser Chrome
Version 71.0.3578.98 (Official Build) (64-bit)
[statisticsreportspictures.zip](https://github.com/MosaicManufacturing/canvas-issues/files/2705268/statisticsreportspictures.zip)

brandonbloch, 2019-01-02:
Hi @rsvp95,

Thank you for the report! However, CANVAS Hub issues are managed separately on the respective plugin repositories. Please open this issue on the Palette 2 Plugin issue tracker: https://gitlab.com/mosaic-mfg/palette-2-plugin

Cheers,
Brandon

closed 2019-01-02 by brandonbloch