MosaicManufacturing/canvas-issues #61: Dec 2018 Keychain doesn't slice correctly with different 1st layer height

joeynuggetz, 2018-12-22:
I imported a S3D profile and my first layer was set to .3 while the rest of the print was .2. The keychain sliced like this:

[image]
[image]

brandonbloch, 2019-01-08:
Hi Joey,

Thank you for pointing this out. After investigating, this seems to be an issue with the model rather than CANVAS, as I see the exact same results with Simplify3D. It looks like that bottom STL is only about 0.13 mm high, which means it gets included at 0.2 mm (it's more than half the thickness of the first layer) but not at 0.3 mm (since it is less than half).

I'll talk to the team about fixing the model so this isn't an issue.

Cheers,
Brandon

brandonbloch, 2019-01-08:
Hi again,

The keychain model files have been updated so that this bottom part is thicker. You should now be able to print at 0.3 mm layers and thicker without it disappearing.

closed 2019-01-08 by brandonbloch