MosaicManufacturing/canvas-issues #62: Canvas

cmbrowning, 2018-12-22:
Canvas when slicing constantly reports server error. Has been occurring at least the last 60 minutes...

SlyFox3007, 2018-12-23:
Yes I second this Server Error.  Running in Chrome, same error but is presented either during slicing or clicking on Preview step.  

Tried to reload page, select different project, or step through the project again.  Seems it ony works if I keep clicking Slice and EVENTUALLY (sometimes 2, sometimes 10, sometimes 20) times it'll go through and slice.

Support Package Attached
[help-package-SlyFox.zip](https://github.com/MosaicManufacturing/canvas-issues/files/2705690/help-package-SlyFox.zip)
[image]

leemun1, 2018-12-23:
@cmbrowning @SlyFox3007 Thank you for reporting this issue. We are aware of this issue and currently investigating into it. While it may be a bit frustrating, as a temporary workaround, we suggest that you try to click the `Slice` button again when you encounter this error. 

We apologize for any inconvenience this issue is causing, and will do our best to resolve it as soon as possible. Thank you!

leemun1, 2018-12-23:
@cmbrowning @SlyFox3007 This issue has been resolved and slicing in CANVAS now works consistently again. Please feel free to re-open the issue if you encounter the same error again. Thank you!

closed 2018-12-23 by leemun1