MosaicManufacturing/canvas-issues #63: Zip or individual STL upload for multicolor object

jamesarm97, 2018-12-23:
Please add support to upload a zip containing multiple STL files for a multicolor part. I tried uploading individual STL files one at a time for a multi color part and it seemed to place them all automatically and did not keep them at object origin. Would like to easily upload from iPad.

brandonbloch, 2019-01-15:
Just want to add a note here: this is planned, but in the meantime you can upload your models individually, select them all in the Object Browser, and click the "Align and Group" button to re-align their origins and group them. This basically has the same effect as a multi-material upload, except that (AFAIK) it doesn't trigger an auto-arrange.

Spooner, 2019-02-23:
This would be great when CANVAS -> Canvas Hub transfer is being flakey, which I find to be half the time, too (to save doing the extract manually)

brandonbloch, 2019-02-25:
@Spooner, OP is referring to model upload rather than sliced file download