MosaicManufacturing/canvas-issues #67: Specific Model Transition Tower

Setesh82, 2019-01-03:
** Problem
Wasted filament due to transition tower.

** Suggestion
Allow a model to be specified for the transition tower instead of the square tower.  This way you would get your piece along with a funky transition colored piece.

mylife4aiurr, 2019-01-23:
Some examples are better
[image]

[image]

[image]

rcmaniac25, 2020-07-15:
One idea that came to mind is if that transition could be treated as a 5th material (similar to how which material is supports can be set to "auto"), this way other objects can be added and be set to the "transition material". Could also be interesting because then you can make certain parts of a model as transition, while others are specific (take the centipede above: what if the body was transition, but the legs/head were specific colors?)