MosaicManufacturing/canvas-issues #71: Side Transistion in Place ends in only Side Transistion

fgnutzi, 2019-01-07:
**Describe the bug**
When choosing 'Sidetransition in Place' in the Printersettings and Setting a Loc for it and then choosing 'Sidetransition' in the Style Settings, the Slicer still shows the Sidetransition with movements in the preview. Open the File in S3D shows that the Printer would not move to the designated Position and would move while transistion. Haven´t tried the file on my Printer as that would cause Problems there. 
Choosing 'no Tansistionhandling' instead end with an Error from the Slicer. 
Choosing 'Transitiontower' get an Transition Tower (as expected)

**Expected behavior**
The Printer moves for Transistion to the , in the Settings designated, Position.

**Operating System (please complete the following information):**
 - Windows 10
 - Browser Google Chrom Version 71.0.3578.98 (Offizieller Build) (64-Bit)

**Additional context**
Side Transition in Place generaly works for me with S3D and Chroma but has some Problems with the Colorchanging. As the colorchanging in Canvas with Towers works very good i would like to try Sidetransition in Place with Canvas.

brandonbloch, 2019-01-08:
Hi @fgnutzi,

Thanks for pointing this out! We just released an update that fixes this issue.

closed 2019-01-08 by brandonbloch

fgnutzi, 2019-01-09:
Hello brandonbloch,

I had the Chance to try the bugfix today but had no success so far. 
The Transition in Place is now shown in the preview and the Printhead moves to the designated Position for the Transistion so far so good. But then he only extrude like 1 cm of Filament and returns to the Object. The Colorchange then took place in the Object.
I had the Transistion Speed set to 2mm/s and the Transitionlength to 160mm, same Settings worked in Chroma.

Also there had been not a single Ping till the 20th Slice (after that i didn´t had the Chance to check for them). But the coloring was out of Calibration till the end of the print.

Thanks for your effort!
Greetings Frank

[Mosaic-Keychain.mcf.zip](https://github.com/MosaicManufacturing/canvas-issues/files/2742022/Mosaic-Keychain.mcf.zip)

brandonbloch, 2019-01-09:
Hi again,

You're absolutely right. After fixing that issue we discovered a math error with splice length calculation when using side transitions. This has now also been fixed, and splices should now be the correct length. Pings also seemed to be back to normal, but do let us know if you're still noticing any pinging issues (and whether you're using connected mode vs. accessory mode).

Thanks,
Brandon

fgnutzi, 2019-01-10:
[201901MosaicKeychain1.zip](https://github.com/MosaicManufacturing/canvas-issues/files/2746347/201901MosaicKeychain1.zip)
Hello Brandon,
i am using connected mode via octoprint.
Had time to test the new Settings but still got an issue with the transistion. It moves to the right spot, extrudes some millimeter Filament and then stays at the spot making a humming noise, still way to short for the Transition. I made a Video, it is in the attachment with the mcf file.
Haven´t looked for pings… as i stopped the print Right after the 1st Transition b/c the noise does not Sound that healthy for my Printer.
Greetings 
Frank

brandonbloch, 2019-01-15:
Hi @fgnutzi,

Looking at the GCode file nothing looks too suspicious. Based on the noise, and how little time the nozzle spends at the side, it may be that your purge speed is too high and your printer is struggling to melt and extrude filament that fast. People in the FB group have reported better results dropping that speed as low as 1 mm/s. It may also be that your printer's firmware has trouble accurately extruding without being synchronized to X/Y/Z movement (Pr3dator, Craftbot's firmware, is one such example) in which case you would need to disable "side transition in place".

Please let us know if either of those helps!

fgnutzi, 2019-01-22:
Hi Brandon,
i didn´t had the chance to try it further as i was upgrading my Printerhardware from 8 to 32 Bit Marlin.But as Sideextrusion worked before when using Chroma and considering that then the only failure was that the Transition unprecise, i raised the current for the feeder to prevent skipping and tried it again with Croma. Even with 2 mm/s the Extruder had no problems pushing the filament out and after raisng the transitionlength to 200mm i could produce good results.
When i find the time again i will try that with Canvas once more.
Greetings Frank

fgnutzi, 2019-01-22:
Hi Brandon,
i didn´t had the chance to try it further as i was upgrading my Printerhardware from 8 to 32 Bit Marlin.But as Sideextrusion worked before when using Chroma and considering that then the only failure was that the Transition was unprecise, i raised the current for the feeder to prevent skipping and tried it again with Croma. Even with 2 mm/s the Extruder had no problems pushing the filament out and after raisng the transitionlength to 200mm i could produce good results.
When i find the time again i will try that with Canvas once more.
Greetings Frank