MosaicManufacturing/canvas-issues #72: Canvas still does not side transition properly

wmmc01, 2019-01-13:
**Describe the bug**
While Canvas now remembers the side transition/purge in place values - it is all messed up. Even before a transition, the print head will move to the transition point, puge a tiny amount, move back to the model, purge a tiny amount, then again back to the purge location, purge a tiny amount, then back to the model.  It purges a tiny amount in total and it is not even at a transition location. It does similar when a transition is reached. 

Current FW & Octo Plugins. 

**Expected behavior**
Proper purging at the defined purge location.

brandonbloch, 2019-01-15:
Hi @wmmc01,

Are you printing in accessory mode or connected mode? If in accessory mode, this is the expected behaviour—these would be the ping sequences in your print, which are inserted during transitions and/or at layer changes, meaning they don't necessarily all line up with transitions. In connected mode,  however, these sequences should never occur so if you're seeing them in connected mode there's definitely something funky going on.

wmmc01, 2019-01-16:
Hello Brandon, Thanks for you prompt reply. I’m only running in connected
mode.

Take care, Bill

brandonbloch, 2019-04-10:
Hi @wmmc01,

So sorry about the delay in responding to this issue. I moved to a new computer a while back and lost some flagged emails, one of which was this message. Can you send the G-code file to brandon@mosaicmfg.com so I can take a look? If this issue is still present we definitely want to get it resolved as quickly as possible.

closed 2019-08-08 by brandonbloch