MosaicManufacturing/canvas-issues #74: canvas online slicing

topautotech1981, 2019-01-20:
*internal server error displayed when attempting to slice any model in all browsers
[image]
 same issue that was stated in canvas#62 by 
cmbrowning and SlyFox3007

topautotech1981, 2019-01-20:
tried clicking slice many times and restarting browser and different models

Madam21, 2019-01-20:
Same, wish Mosaic would comment on this issue. Sucks having a expensive paper weight....

brandonbloch, 2019-01-22:
Hi everyone,

Our apologies about this. Some of our servers went down for about a 24 hour period over the weekend, resulting in some users being unable to slice entirely. We understand that there's nothing more frustrating than encountering a problem you can't work around in any way. Please know that this issue is a top priority for us, and we're working on resolving these issues entirely so that the slicing process does not experience this type of outage in the future (as well as being faster overall).

We have brought the servers back online, so you should now be able to slice once again. I'll leave this issue up for now so that others can follow along here. Please let us know if you have any questions or if you're still experiencing this issue.

Cheers,
Brandon

closed 2019-03-06 by brandonbloch

r3df0xd13, 2021-02-01:
Having similar issue now. When slicing my model, Canvas will reach 1%-8% depending on the size of the model and then generate a Server Timed Out While Slicing error message