MosaicManufacturing/canvas-issues #76: Unexpected "An error ocurred while auto-arranging"

cwvhogue, 2019-01-22:
Problem:
Hit "An error ocurred while auto-arranging" with 4 STL files loaded individually and grouped in the Object Browser.  

Files are sucessfully uploaded, and processed all the way to completed slicing.  The grouping on the Object Browser is done with the link icon at bottom.  Clicking the ... menu (on left side) and the "Auto Arrange" button produces the following:

[image]

STL is deposited in: 
https://www.thingiverse.com/thing:3372731

**To Reproduce**
1. Download the 4 STL files. 3CRO_A-export, 3CRO_B-export, 3CRO-L-export, 3CRO-R-export
2. In Canvas, use the Left Side "Multi-Material" upload button to upload these one at a time.
3. In the Object Browser, select all 4 files and click the "link" icon at the bottom to group the files.
4. From the ... menu on the left side, click Auto Arrange.
5. See error

**Expected behavior**
Everything works on these 4 files, they are together larger than the 100Mb upload maxium size, but can be uploaded individually and grouped. They can be sliced in the position they are in, resulting in a printable file. 
 
Only Auto Arrange fails, which should properly line up the 4 parts, making a DNA double helix with 2 bound proteins. Again these 4 STL files have the same XYZ origin and can be correctly arranged in other software (e.g. IdeaMaker).   To help you see how they should be arranged:

[image]

**Operating System:**
 - OS: Windows 10
 - Chrome  71.0.3578.98 (Official Build) (64-bit)

brandonbloch, 2019-01-23:
Hi @cwvhogue,

Sorry you've run into this issue. Our slicing servers have been experiencing intermittent load issues over the past week or so, and auto-arrange currently runs on these servers. They are currently up and running so this should not be a problem at the moment. While we are working on improving reliability and performance of these servers, we're also working on a faster approach to auto-arrange that should run very quickly. Assuming all goes well, this should ship in the next few weeks, which will both improve the auto-arrange experience and alleviate some of the load on the servers.

As for fixing the alignment of these models, there is a lesser-known alternative workflow that can do this without auto-arrange. If you ungroup those models, and then click the Align and Group button, it will align the models with one another based on their origins. From there you can manually position the group as desired. Hopefully this helps in the meantime!

[image]

Cheers,
Brandon

cwvhogue, 2019-01-26:
Thanks, this worked!

closed 2019-01-26 by cwvhogue

brandonbloch, 2019-01-29:
Hi again,

Just following up to let you know that the new auto-arrange functionality has shipped. The operation should now be (for all intents and purposes) instant! Let us know if you experience any issues.

Cheers,
Brandon