MosaicManufacturing/canvas-issues #79: Support is current filament

mylife4aiurr, 2019-01-23:
[image]
Currently supports are tied to a specific extruder. So for instance a 4 color print. User must assign an extruder for support. If the user chooses support extruder as 3. Then every color that isnt extruder 3 needs a splice created and then extra time and material wasted to get back to support extruder 3's color. So extra splice and extra time needed to transition back to the support extruder.
[image]
Why not make the support extruder the current color. Now a new splice isnt necessary, which also saves time because a transition isnt necessary. 
Of course this would not apply the something like PVA supports or interface support only, in this case you want a specific extruder for support only. This request is only same material as support.

brandonbloch, 2019-01-23:
Thank you for this request! This is definitely the ideal situation (and the same would apply to rafts and skirts/brims) and is planned in the medium-term. The default option would basically be "auto" with the ability to force a specific extruder as is the current behaviour, for printing with solubles.

mylife4aiurr, 2019-01-23:
Ok thanks 4 acknowledgement. just looking to save time and splices

mylife4aiurr, 2019-07-07:
Hey brandon, 
some1 else just asked a very similar question as facebook group
https://www.facebook.com/groups/mosaic.users/permalink/391412704828637/

brandonbloch, 2020-03-31:
Hi all,

I know it's been a while since this was requested, but this feature is now here with the option to select "Auto" for the support extruder. This will make sure to re-use an extruder on that layer, so as to not add a transition. This does not affect interface layers if enabled.

Cheers,
Brandon

closed 2020-03-31 by brandonbloch

mylife4aiurr, 2020-03-31:
Thank you for your work Sir 👍