MosaicManufacturing/canvas-issues #81: Support Block/Enforcement

dbsweeny, 2019-02-08:
I would find it useful (as I imagine other users would, too) to be able to either enforce and/or block supports being autogenerated in specific locations on a model. I'll spare you too much description of what I mean as this is a common feature on other slicing software, so I'm sure you're aware of the functions of the feature, but Canvas often generates supports where I understand they technically should be, but where I know from experience they are not actually needed, so I would like to be able to block some, but not all, supports in order to speed up a print and reduce its waste. Super useful.

Spooner, 2019-02-23:
Usually there is a "only supports touching bed" and "supports everywhere" option (well, in Cura there is :D)

dbsweeny, 2019-02-23:
An example from Slic3rPE of a custom shape (the green cube) that can be set as either a support enforcer or support blocker. This is the sort of thing I had in mind.

[image]

Spooner, 2019-02-23:
Ah sorry, that is clear now (and obviously useful!)

brandonbloch, 2020-03-31:
Hi all,

This feature is finally here! With the new support painting UI, you can manually designate areas for support, starting from a blank slate or an automatic fill that can do a "from bed only" style fill!

Cheers,
Brandon

closed 2020-03-31 by brandonbloch