MosaicManufacturing/canvas-issues #82: Side Transition support for Marlin based Printers using software endstops

jtkdaddy, 2019-02-11:
**Is your feature request related to a problem? Please describe.**
Side transitions are not working with my CR10S4 flashed with TH3DStudio firmware (Marlin 1.1.9 based) due to software endstops being enforced.

**Describe the solution you'd like**
Either
a) A checkbox option that when enabled, would override software endstops in Canvas/Chroma which would issue a M211 S0 at the start of side transitioning and M211 S1 at the end of side transitioning.
or 
b) Start/End script sequence command override text boxes in Canvas/Chroma that would be executed at the start of a side transition and also at the end.  Maybe even make it more generic and allow for custom script at beginning/end of any type of transition.

**Describe alternatives you've considered**
Executing M211 S0 at the start of a print job.  However, this is non ideal since software endstops would be disabled for the entirety of the print job.

**Additional context**
Additional info can be found here http://marlinfw.org/docs/gcode/M211.html.

brandonbloch, 2019-08-08:
Hi @jtkdaddy,

The ability to insert custom sequences before and after side transitioning is here! You can edit these sequences when editing your printer settings, in the Sequences tab. This gives a large amount of flexibility in terms of motion and firmware-specific commands such as `M211`, and so should support your use case and more!

If all you need is to disable and re-enable software endstops, I'd recommend a usage along the lines of the following:

**Before side transitioning**

```
M211 S0 ; disable software endstops
```

**After side transitioning**

This depends on whether you need to move inside the build volume to re-enable endstops. If not, you can re-enable them while still outside then `M211 S1` should be all you need. I assume you need to, so I'd recommend something like this:

```
G1 X<NEXTX> Y<NEXTY> F[FEEDRATE] ; where [FEEDRATE] is your XY travel speed
M211 S1 ; re-enable software endstops
```

Cheers,
Brandon

closed 2019-08-08 by brandonbloch

jtkdaddy, 2019-08-08:
Awesome!  Thanks and much appreciated.