MosaicManufacturing/canvas-issues #83: Internal Server Error (2012) on using duplicate colours

Spooner, 2019-02-16:
**Describe the bug**
When using the same colour, even on unused extruders, I get a 2012 error when I go to the preview page. This happened when I used only ex1&2 in the model and had the same colour set on ex 3&4.

**To Reproduce**
Set extruder 1 => Grey
Set extruder 2 => Yellow
Set extruder 3 => Purple
Set extruder 4 => Purple

go to preview page

**Expected behavior**
Either:

* should give me a meaningful error message: "All 4 extruder colours must be unique"
* doesn't care because the colours are on unused extruders

**Operating System (please complete the following information):**
 - OS: Ubuntu 18.04
 - Browser: Chrome
 - Version: Tested on 2018-02-16

Spooner, 2019-02-16:
Not sure how this works. Doesn't seem to actually be based on same colours. May be something about using (or not using) the material-specific setting (I have each extruder set at a different temp). Or maybe it is just being "funny today"?

Spooner, 2019-02-17:
Yeah, definitely an issue with trying to define a non-default material that changes temperature from default. You can define such a material, but if you use it, you get 2012 on going to slice.

brandonbloch, 2019-02-19:
Hi @Spooner,

I just attempted to re-create this issue and could not get the error to occur. I'm thinking it might have to do with something more specific to your project or material profiles. Do you mind sending your support bundle to canvas@mosaicmfg.com so we can take a closer look?

Thanks,
Brandon

Spooner, 2019-02-19:
My initial report was faulty, sorry. It seems that the cause is a particular material being applied:
[image]

Removing the "/" from the name of the material seems to fix it...

brandonbloch, 2019-02-20:
Hi @Spooner,

Thanks for the additional info! I've confirmed and fixed this issue. You may now use slashes to your heart's content!

closed 2019-02-20 by brandonbloch