MosaicManufacturing/canvas-issues #84: Small print with just a few layers cause "no ping" error before the filament has even reached the printer

jorgtheelder, 2019-02-17:
**Describe the bug**
When I print small/thin things meant to be flexible, I get "no ping detected" before it has even spliced enough filament to reach the printer. Very annoying.

**To Reproduce**
Steps to reproduce the behavior:
1. Make a small two color item that only has 6 or so 0.2mm layers.
2. Try to print

**Expected behavior**
It should never try to detect pings when the print has not even started on the printer side.

**Operating System (please complete the following information):**
Does not matter. Happens even when loading from SD.

**Additional context**
A sample that caused the issue is attached.

brandonbloch, 2019-02-19:
Hi @jorgtheelder,

This sounds like a Palette 2 firmware bug rather than a CANVAS bug. I'll pass this on to the correct team but would generally recommend posting directly in the Palette 2 support forums: https://support.mosaicmfg.com/hc/en-us/community/topics/360000498854-Palette-2-Palette-2-Pro

Also please note, we released v1.1.1 of the firmware updater last week. However, this is just the updater application itself, not the firmware version. The current version of Palette 2 firmware is 4.0.3, released on Thursday. You can check the device's version directly by going to Settings > About and reading the version from there. I just want to confirm that you're on the latest version as this may be a bug that's already been fixed.

Cheers,
Brandon

brandonbloch, 2019-02-19:
Hi again,

I've just spoken with the team. They've logged the issue and will make sure to look into it. If you'd like more information please feel free to get in touch by email at support@mosaicmfg.com.

Closing this issue as it isn't directly related to CANVAS functionality.

closed 2019-02-19 by brandonbloch