MosaicManufacturing/canvas-issues #85: Actively support touch controls

Spooner, 2019-02-19:
I can use most, but not all, of the webpage with my tablet.

The only thing that I can find which is impossible is to be able to drag colours from the dots onto models (some other bits are fiddly, but nothing else is actually impossible)

P2 and printer are in a different room than my desktop PC and it would be great to be able to use my tablet to control canvas when in my workshop (or without having to boot up my PC).

brandonbloch, 2019-02-20:
🎉

closed 2019-02-20 by brandonbloch

Spooner, 2019-02-20:
Thanks - so quick! However, now it is very difficult to open/close the
materials window, without just changing the colour of what you last painted.

Spooner, 2019-02-20:
Oops, yes, if you touch the white border around the paint colour, the material window toggles. Touch the colour and you set that colour.

brandonbloch, 2019-02-21:
Hmm, sounds like I spoke a bit soon. This was working during testing but some further optimizations were required and it seems they have broken this aspect of the interaction. I've dug into why and it turns out that trying to fix it has unfortunately made things worse, so we're going to leave this behaviour as-is for now.

Please note that proper mobile/touch support is on our roadmap and we will be properly re-working this interaction (along with optimizations of the general layout for small screens) at that time. In the meantime, I hope this level of support will suffice!

Cheers,
Brandon

Spooner, 2019-02-21:
It works now, even if it is a bit fiddly to use. I am happy :)