MosaicManufacturing/canvas-issues #87: 2020 error on slicing

Spooner, 2019-02-21:
**Describe the bug**
Internal server error on preview page.

*To Reproduce**
import 4 parts of earbud and give them different colours (doesn't care which materials I select)
go to preview page
popup "Internal Server Error 2020"

Specific failing project: https://canvas3d.io/slicer/3c8038b80a4e1770280bab580681e16e/preview

**Operating System (please complete the following information):**
 - OS: Android
 - Browser chrome
 - Version 2018-02-21

Spooner, 2019-02-21:
You __really__ need to add some text to your error messages - would be a lot easier for us to diagnose what was really going wrong (and manage a workaround at our end, until the problem is fixed).

brandonbloch, 2019-02-22:
Hi @Spooner,

I believe this issue was caused by a bug with detection of ASCII vs. binary STLs which we've resolved this afternoon. Can you please confirm this?

And yeah, unfortunately they’re not too informative for internal server errors. All other types of errors include a description of the error. As the application matures we’re working to differentiate more errors from internal server errors so that we can provide these descriptions.

Not providing descriptions for internal server errors was a conscious decision because these types of errors are, by nature, unexpected within the system, and exposing the causes of these errors can introduce potential security risks. Including messages for some but not all these errors would still indicate that the nondescript, cryptic errors are related to security or private data and thus provide “too much” information for malicious use. I hope that justification makes sense, and know that the inclusion of the numbers has been helping us eliminate the issues that cause these errors fairly quickly.

Cheers,
Brandon

Spooner, 2019-02-22:
Yeah, that project slices again now :)

brandonbloch, 2019-02-22:
Glad to hear! Have a great weekend and happy printing :)

closed 2019-02-22 by brandonbloch