MosaicManufacturing/canvas-issues #88: Have control over auto-arranging

Spooner, 2019-02-23:
**Is your feature request related to a problem? Please describe.**
I want to print lots of small things close together, with a shared brim. If I arrange them all manually, it takes a while, then if I want to add or remove an item, then it auto-arranges and makes me start again!

Also, auto-arrange tends to put things in two completely separate blocks, which confuses me.

**Describe the solution you'd like**
Auto-arrange on/off and/or auto-arrange min-distance settings.

**Additional context**
Manually laid out (one at a time!). I ideally want to be printing 100 things at once which would be painful to manually set!

[image]

Spooner, 2019-02-23:
Here is auto-arrange, clearly picking two separate areas to use. Why is there a big gap in the bottom left? Is it possible that is a gap left for the transition tower (the tower could fit in that gap, but it it entirely on the left side of the used areas).

[image]

Spooner, 2019-02-24:
In the end, I used 10 brim lines to make the auto-arranged models share brims and therefore adherence strength. If I could make them arrange a bit closer, I could get that down to 4 lines, which saves a fair bit of time and marerial.

bash-t, 2020-04-23:
Hi,
I would also really appreciate it if there would be a possibility to turn off auto arrangement.
I often have buildplates arranged very carefully manually, since I wanted to print dozens of miniatures on them in one go to only have one waste tower at a time. To help stick them better to the bed, I always move them quite close together and add a brim around them to kind of fuse them together on the first layer. However, after spending a long time arranging everything, and then wanted to decide to have another copy of one element - boom, the auto-arrangement instantly made all your flimsy manual arrangement useless. Even more annoying since there is no "undo" function, you'll have to start all over again.

An option to disable the auto arrangement would be highly appreciated.

Many thanks!
[image]

accetturo, 2020-06-01:
I have this issue as well and it's a nightmare trying to align files and costs me at least a few hours a week.. It makes me dread using this slicer. We also need better duplication control, aka duplicate 20 items instead of 1 by 1

brandonbloch, 2024-07-08:
Hi all,

It's been in development for a while, but this morning we released a brand-new auto-arrange algorithm! Some key highlights:

- Allows better model tesselation, rather than only arranging by bounding box
- Allows for automatic rotation of models about the Z axis (allowed rotations can be configured)
- Allows spacing between models to be configured
- When uploading or duplicating models, existing models will be left in their current positions while new models will be auto-arranged around them

We hope that this addresses all of the shortcomings of the previous auto-arrange algorithm, and makes working with Canvas even easier!

Happy printing,
Brandon

closed 2024-07-08 by brandonbloch