MosaicManufacturing/canvas-issues #89: gcode file transferred but not saved

Spooner, 2019-02-24:
**Describe the bug**
when sending a gcode file from CANVAS to CANVAS hub, it will transfer correctly (the progress bar shows and goes from 0 to 100% OK). However, then it shows the "File received, please wait" popup that never goes away and the file has not been saved to the Octoprint machine.

*To Reproduce**
Sometimes, I can just send a gcode from CANVAS to CANVAS Hub without problems. Other times

**Operating System (please complete the following information):**
 - OS: Android or Ubuntu
 - Browser  chrome

leemun1, 2019-02-25:
@Spooner Hi, thank you for submitting this issue! This seems to be more related to the CANVAS Octoprint plugin, rather than an issue with CANVAS itself. I have notified the CANVAS Hub team about this issue. If you would like you can also post an issue on their [issue tracker](https://gitlab.com/mosaic-mfg/canvas-plugin/issues). 

In the meantime, we would like to ask you for some additional information surrounding your issue:

1. Does this bug happen more often with certain types of gcode files? (e.g: large files vs small files? single-color vs multi-color?)
2. When you say it is not saved to CANVAS Hub, do you mean you are unable to find the file using the search bar in OctoPrint?
3. Are you using a CANVAS Hub or your own Raspberry Pi with the CANVAS Plugin?

Spooner, 2019-02-25:
1. Only noticed the problem with sending multi-material gcode through (though that is the only type of gcode I've been making with CANVAS recently). I am working with small two-colour items, with 8-40 items on the bed, at the moment. 3-5MB files. Not noticed ant correlations there.
2. I can't see it in Octoprint's files list, even when I refresh the list or whole web page.
3. rPi3B + Rasbian + Octoprint

leemun1, 2019-02-25:
@Spooner Thank you for the quick reply. Could you please send your octoprint.log to support@mosaicmanufacturing.com and link this thread in your e-mail? We’ll definitely keep an eye on this issue. If this issue happens again, please send your most recent octoprint.log to the same e-mail as above as soon as it happens - it will help us greatly in troubleshooting.

Spooner, 2019-02-25:
Here are all my logs for a while: [octoprint.logs.zip](https://github.com/MosaicManufacturing/canvas-issues/files/2902675/octoprint.logs.zip)

Files have transferred OK today, but have failed in the last few days. You might find it hard to find failures in all that, so I'll upload my latest log when the problem recurs.

Spooner, 2019-03-11:
Still getting this issue. Interestingly, if the "File received, please wait" popup is showing and you upload manually, it disappears and shows that the file was added.

osubuu, 2019-03-11:
@Spooner Hi,

Could you please upload your latest octoprint.log when you can? We'll look into your recent octoprint history and try to catch the bug that is causing this.

Spooner, 2019-03-11:
@osubuu It is doing it now, but I'm in the middle of a print, so it is pretty spammy in there. I will clear the log and just attempt the send once it is finished :)

Spooner, 2019-03-11:
@osubuu 
```
2019-03-11 23:15:53,760 - octoprint.plugins.canvas - INFO - RECEIVED DELTA
2019-03-11 23:15:53,761 - octoprint.plugins.canvas - INFO - Payload: {"version":356,"timestamp":1552346153,"state":{"queuedPrint":{"userId":"c3db214ee861125d083c871316da4b9a","projectId":"70201ee2939b5a2e2b092f8ad7312ea2","filename":"Keyforge-Stun-x40"}},"metadata":{"queuedPrint":{"userId":{"timestamp":1552346153},"projectId":{"timestamp":1552346153},"filename":{"timestamp":1552346153}}}}
2019-03-11 23:15:53,767 - octoprint.plugins.canvas - INFO - Status: delta/canvas-hub-477fc0b5159292f2b5aef275d24f70c1
2019-03-11 23:15:53,769 - octoprint.plugins.canvas - INFO - Token: None
2019-03-11 23:15:53,770 - octoprint.plugins.canvas - INFO - Handling Prints
2019-03-11 23:15:53,774 - octoprint.plugins.canvas - INFO - {u'queuedPrint': {u'projectId': u'70201ee2939b5a2e2b092f8ad7312ea2', u'userId': u'c3db214ee861125d083c871316da4b9a', u'filename': u'Keyforge-Stun-x40'}}
2019-03-11 23:15:54,670 - octoprint.plugins.canvas - INFO - Sending UIUpdate from Canvas
...lots of Sending UIUpdate from Canvas...
2019-03-11 23:15:55,287 - octoprint.plugins.canvas - INFO - Sending UIUpdate from Canvas
2019-03-11 23:15:55,560 - octoprint.plugins.canvas - INFO - SHADOW UPDATE RESPONSE
2019-03-11 23:15:55,560 - octoprint.plugins.canvas - INFO - Payload: {"state":{"reported":{"queuedPrint":null},"desired":{"queuedPrint":null}},"metadata":{"reported":{"queuedPrint":{"timestamp":1552346155}},"desired":{"queuedPrint":{"timestamp":1552346155}}},"version":357,"timestamp":1552346155,"clientToken":"46c4b0b2-38af-4d75-b5c7-5f5d42b50dc1"}
2019-03-11 23:15:55,561 - octoprint.plugins.canvas - INFO - Status: accepted
2019-03-11 23:15:55,564 - octoprint.plugins.canvas - INFO - Token: 46c4b0b2-38af-4d75-b5c7-5f5d42b50dc1
```

osubuu, 2019-03-12:
@Spooner, could you try enabling active folder polling (see screenshot), press SAVE, and then try sending a file from CANVAS to your Hub again?
[image]

Spooner, 2019-03-12:
@osubuu Turned on rPi today and it wasn't transferring again. Clicked that setting and saved...and still not working, sorry! Even restarting octoprint didn't help.

osubuu, 2019-03-13:
@Spooner That's odd. Could you connect remotely into your Pi (SSH) and monitor the contents of your OctoPrint "watched" folder? The path for the "watched" directory is indicated in the same location as the active folder polling option above. 

Once you are in your Hub, try sending a file from CANVAS to your Hub again and then once the notification reaches the "File received. Please wait..." status, please check if the .gcode file was added as a file in the "watched" folder directory.

If you need any help with any of the steps above, we can give more detailed steps.

Spooner, 2019-03-13:
@osubuu (I'm a dev, so ssh no problem)
Hmm, the watched/ folder is full of files (uploads/ is full of the files I can actually see in the web interface)! So I am guessing that all transfers of the files that are already in watched/ aren't getting detected since they don't create new files, just replace them?

```
pi@octopus:~/.octoprint $ ls watched/
201901MosaicKeychain1.mcf.gcode    Keyforge-Card-box-unsleeved.gcode
3DBenchy.gcode                     Keyforge-damage-5x2-3x6-1x14.mcf.gcode
four-square-retraction-test.gcode  Keyforge-power-+3x1-+1x9.mcf.gcode
Keyforge-Aember-1x49.gcode         Keyforge-Stun-x40.mcf.gcode
Keyforge-Armour-x50.mcf.gcode
pi@octopus:~/.octoprint $ ls uploads/
arkham_horror  dice_hospital  eclipse                  keyforge    tools
azul           dice_trays     folklore_the_affliction  token_tray
pi@octopus:~/.octoprint $ 

```

Spooner, 2019-03-13:
Just tried transferring a file - with the same name as one in watched/. Never gets fully transferred. Deleted the files in watched/ and tried again and it transferred file. At least I know how to fix it now, but the question is why am I getting watched clogged up with files?

osubuu, 2019-03-13:
@Spooner For some reason, OctoPrint doesn't seem to be picking up changes to your watched folder correctly. Theoretically, the watched folder should always be empty if the files added to it are recognized by OctoPrint (i.e .stl, .gcode, any 3D-printing-related extensions). Your files all seem to have proper extensions, so I find it weird that this isn't happening.

The "File Received. Please wait..." popup only shows up when a file is put into the watched folder. The popup will change to "CANVAS File Added" when OctoPrint analyzes it and transfers it to the uploads folder correctly.

Although we haven't been able to re-create this issue on our end, we'll keep monitoring it to hopefully find the cause of why OctoPrint isn't behaving as expected. Sorry for the trouble, please let us know if you discover anything else!

closed 2019-07-04 by brandonbloch