MosaicManufacturing/canvas-issues #90: Print Summary 2 colors, maf file 3 colors

hojalot, 2019-02-24:
I uploaded a 2 part test pattern.  In the "Color" Canvas step, I hide each part, drag and drop a color on it, show both, then "Preview" to slice.  The "Print Preview" window shows the 2 colors "White" and "Yellow" I chose for Drive 3 and Drive 4.

Size: 55 x 54.7 x 8.9 mm
Time: 2 hr 27 min
Filament Length: 7.42 m
  [white] Drive 3:    3.51 m
  [yellow] Drive 4:    3.92 m
Palette
  Total Splices:  41
  Total Pings:    19

When I download the .maf file, it contains a third unexpected "Blue":
O25 D13c71c1Blue_Filament D0 D1ffffffWhite_Filament D1fcca46Yellow_Filament

When I start the print in Pallete2, it prompts to load 3 filament colors when only 2 were expected.

brandonbloch, 2019-02-25:
Hi @hojalot,

I have a suspicion that this is caused by a purge in your start sequence. Can you share your starting GCode so we can see?

hojalot, 2019-02-26:
The "Start sequence" in Canvas Printer Profile is just "G92 E0".  I set the bed and extruder temperatures manually.

brandonbloch, 2019-02-27:
I'm unable to reproduce this with my printer profiles. Can you send the .maf and .gcode files to support@mosaicmfg.com so we can see if there's anything else causing this issue?