MosaicManufacturing/canvas-issues #91: Inconsistent confirmation of deletion

Spooner, 2019-02-25:
**Describe the bug**
Sometimes we are asked to confirm deletion of models. Sometimes we don't.

**To Reproduce**
Select model & click deletion icom => confirmation dialog appears before deletion.
Click delete icon => Immediate deletion.

**Expected behavior**
Since there is no UNDO functionality (which is terrible in itself) confirmation should always be required.

**Operating System (please complete the following information):**
 - OS: Ubuntu/Android
 - Browser chrome

brandonbloch, 2019-02-25:
I believe this has to do with deleting a single model vs deleting all selected models (or a group). The confirmation modal isn't too disruptive so I think showing it even for single deletion makes sense.

Undo/redo functionality is planned; however, undoing the deletion of a file becomes complicated quickly so it may fall outside the scope of that functionality. We'll have to figure out if we can implement it in a way that is easy to use but isn't wasteful.

Spooner, 2019-02-25:
No, I'm talking specifically about deleting a group, using the icon button. Behaviour is different whether the group is selected or not.

brandonbloch, 2019-02-25:
Ah, thanks for the clarification.

brandonbloch, 2019-08-29:
Our apologies—this issue has been resolved but we haven't closed it. Model deletion always requires a confirmation, regardless of the number of models or whether they're grouped or not.

closed 2019-08-29 by brandonbloch