MosaicManufacturing/canvas-issues #93: Not handling retract / unretract from transition tower correctly

cmroche, 2019-03-03:
**Describe the bug**

I email support about a similar issue happening with Chroma. 

When producing a GCODE with the web slicer, a retraction is generated before the move to the tower however a recovery is not generated before the tower starts. This is visible in the tower prints as a missed extrusion as the start of the purge.

Once the purge tower layer is complete there is a missing retract in the GCODE prior to moving back to the part, the move back to the part is completed, then a retract recovery which causes a large blob to be deposited. This is probably more noticeable for those using bowden extuders.

**To Reproduce**

Just slice something with retraction enabled and look at the gcode generated for the purge tower.

**Expected behavior**

Retraction prior to purge tower
Move to purge tower
Retract recovery
Print purge tower
Retraction prior to move to part
Move back to part
Retract recovery
Print part

**Screenshots**

Refer to issues #92 for screenshots of my configuration.

**Operating System (please complete the following information):**
 - OS: OSX 10.14.3
 - Browser Chrome and Safari
 - Version Latest

**Additional context**
This is nearly identical to a problem with purge towers being generated in Chroma.

cmroche, 2019-03-03:
For reference, this is the GCODE being generated

```
; extruder(s) off
M103
;
; 'Crown Path', 0.0 [feed mm/s], 2.5 [head mm/s]
G1 X154.233 Y135.998 E0 F9000
; extruder on
M101
; head speed 2.499990, filament speed 0.040328, preload 0.000000
; Prepare for End-Of-Layer
G1 X154.161 Y136.206 E0.00355 F150
G1 X154.04 Y136.46 E0.00454
G1 X153.886 Y136.679 E0.00432
G1 X153.627 Y136.764 E0.0044
; extruder(s) off
M103
;
; 'Destring/Wipe/Jump Path', 0.0 [feed mm/s], 4.9 [head mm/s]
; 'Destring Suck'
G1 E-3 F3300
; head speed 4.900000, filament speed 0.000000, preload 0.000000
; Prepare for End-Of-Layer
;
; *** Cooling Extruder 2 to 210 C ***
;
;
;
; *** Selecting and Warming Extruder 3 to 210 C ***
M104 S210 T0
; Chroma toolchange handling
; toolchange at totalE = 16661.99881
; changing logical extruder from T1 to T2
; chroma inserting next dense tower segment
; move to transition tower
G1 X100.903 Y120.049 F9000
G1 Z28 F1800
; extruder on
M101
G1 X100.766 Y119.912 E0.00626 F6000
G1 X100.492 Y119.637 E0.01251 F6000
G1 X100.354 Y119.5 E0.00626 F6000
G1 X99.806 Y119.5 F9000
G1 X99.928 Y119.622 E0.00556 F6000
G1 X100.293 Y119.988 E0.01669 F6000
G1 X100.415 Y120.11 E0.00556 F6000
G1 X100.781 Y120.475 E0.01669 F6000
G1 X100.903 Y120.597 E0.00557 F6000
G1 X100.903 Y121.146 F9000
G1 X100.8 Y121.043 E0.00469 F6000
G1 X100.492 Y120.735 E0.01408 F6000
G1 X99.668 Y119.912 E0.03755 F6000
G1 X99.36 Y119.603 E0.01408 F6000
G1 X99.257 Y119.5 E0.00469 F6000
G1 X98.708 Y119.5 F9000
G1 X98.845 Y119.637 E0.00626 F6000
G1 X99.257 Y120.049 E0.01878 F6000
G1 X100.354 Y121.146 E0.05006 F6000
G1 X100.766 Y121.558 E0.01878 F6000
G1 X100.903 Y121.695 E0.00627 F6000
G1 X100.903 Y122.244 F9000
G1 X100.793 Y122.134 E0.00501 F6000
G1 X100.464 Y121.805 E0.01502 F6000
G1 X99.915 Y121.256 E0.02504 F6000
G1 X99.147 Y120.488 E0.03505 F6000
G1 X98.598 Y119.939 E0.02504 F6000
G1 X98.269 Y119.61 E0.01502 F6000
G1 X98.159 Y119.5 E0.00501 F6000
G1 X97.611 Y119.5 F9000
[... 1726 lines cut ...]
G1 X72.006 Y164.68 E0.01402 F6000
G1 X71.903 Y164.578 E0.00467 F6000
G1 X71.903 Y165.127 F9000
G1 X71.996 Y165.22 E0.00426 F6000
G1 X72.183 Y165.407 E0.00852 F6000
G1 X72.277 Y165.5 E0.00426 F6000
; leave transition tower
; extruder(s) off
M103
G1 Z28 F1800
; endchroma
;
;
; *** Estimate 7.235519 [s], Slowing (0.140x) to 20.0 [s] ***
; Perimeter Length = 14.792214
;
; 'Perimeter Path', 0.2 [feed mm/s], 5.6 [head mm/s]
G1 X136.078 Y149.466 Z28 E0 F9000
; 'Destring Prime'
G1 E3 F3300
; extruder on
M101
; head speed 5.599999, filament speed 0.180670, preload 0.000000
G1 X135.997 Y149.462 E0.00263 F336
```

brandonbloch, 2019-03-06:
Hi @cmroche, I believe this has to do with #92 (a legacy firmware that our slicing engine supports uses M101/M103 commands, but we don't support these in CANVAS—the fact that they're showing up was part of that bug). The transition tower logic was inserting these M10x commands instead of retractions. I believe that the fix for #92 also addresses this; please feel free to re-open this issue if that fix does not also address this.

closed 2019-03-06 by brandonbloch

cmroche, 2019-03-06:
I will check tonight and let you know. Thanks @brandonbloch

brandonbloch, 2019-03-06:
Yes, the fix has shipped. If you re-slice your project you should see the G10/G11 commands as expected.