MosaicManufacturing/canvas-issues #95: Firmware retractions are being double up

cmroche, 2019-03-06:
**Describe the bug**

When firmware retractions are enabled, the correct G10 and G11 sequences are generated, but corresponding G1 commands are also present doubling them up (only on the prime tower, other uses are correct)

```
G1 X170.247 Y106.475 E0.01068
G1 X170.176 Y106.559 E0.00356
; trigger retract
G10
;
; 'Destring/Wipe/Jump Path', 0.0 [feed mm/s], 20.0 [head mm/s]
; 'Destring Suck'
G1 E-3 F3300
; head speed 20.000000, filament speed 0.000000, preload 0.000000
; Perimeter Length = 50.452721
;
; 'Perimeter Path', 0.6 [feed mm/s], 20.0 [head mm/s]
G1 X190.787 Y84.626 E0 F9000
; 'Destring Prime'
G1 E3 F3300
; trigger prime
G11
; head speed 20.000000, filament speed 0.645247, preload 0.000000
G1 X190.732 Y84.608 E0.00188 F1200
```

**To Reproduce**

Enable firmware retractions for your printer
Slice any model
Inspect the GCODE search for "G10" and locate a retraction to and from the prime tower

**Expected behavior**

G1 E commands are omitted, as they are in other retraction cases.

**Screenshots**

See #93 for screenshots of my setup

**Operating System (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
This is related to #93 and #94 which otherwise appear to be fixed outside of this problem/

Thanks @brandonbloch

cmroche, 2019-03-16:
I was able to come up with a temporary work around, I set the retraction distance to .00000001 which generates `G1 E0 F3300` for both the retracts and detracts on the tower. My firmware seems more than happy to ignore these. Just in case anyone else is looking for a work around in the interim.

brandonbloch, 2019-03-21:
Hi @cmroche,

Apologies about the delay in getting to this. I'm glad you were able to work around this! However, this issue has now been resolved and the fix has been pushed out. I also found that the print preview wasn't showing firmware retracts/restarts so that's been fixed as well. Thanks for being so thorough with this report, and please let us know if there's still anything funky!

Cheers,
Brandon

cmroche, 2019-04-10:
@brandonbloch It's been a couple of weeks since I've been able to confirm, but I am definitely still seeing double retractions in there.

This is the GCODE I'm seeing from a slice I made tonight of the Jan 2019 keychain

'''
; trigger retract
G10
;
; 'Destring/Wipe/Jump Path', 0.0 [feed mm/s], 20.0 [head mm/s]
; 'Destring Suck'
G1 E-2.5 F3300
; head speed 20.000000, filament speed 0.000000, preload 0.000000
; Perimeter Length = 29.434536
;
; 'Perimeter Path', 0.7 [feed mm/s], 20.0 [head mm/s]
G1 X167.206 Y116.702 E0 F9000
; 'Destring Prime'
G1 E2.5 F3300
; trigger prime
G11
; head speed 20.000000, filament speed 0.665203, preload 0.000000
'''

The workaround does still work to make them effectively 0 length.

When search through the output gcode files, keep in mind that this only applies to retractions for the wipe tower, all other retractions in the gcode are using G10/G11 without the G1 retracts as would be expected.

brandonbloch, 2019-04-10:
Well this is weird. It was definitely fixed and I've tested it just now and it seems to still be working on my end. Specifically, the change sets your retract length to 0 if firmware retractions are enabled. Would you be willing to send your login info and the name of the project to brandon@mosaicmfg.com so I can see why this is happening only in your case? (If so, please change your password beforehand!)

closed 2019-04-11 by brandonbloch