MosaicManufacturing/canvas-issues #96: Incorrect rendering of specific STL files (no lighting)

Spooner, 2019-03-20:
**Describe the bug**
Rendering of specific stls shows them without any lighting. I'm guessing there are no normals in the stl, but shouldn't you guess at them? STL viewer show the object properly.

**To Reproduce**
Load any of the stls in
[7_Wonders_+_Expansions_storage_solution_.zip](https://github.com/MosaicManufacturing/canvas-issues/files/2988077/7_Wonders_%2B_Expansions_storage_solution_.zip)
(from [thingiverse](https://www.thingiverse.com/thing:3293262))

**Expected behavior**
Should show the model lit up so you can see the geometry properly.

**Screenshots**
[image]

[image]

**Operating System (please complete the following information):**
 - OS: Android 8.1.0
 - Browser chrome

Spooner, 2019-03-20:
p.s. slicing seems to work correctly

Spooner, 2019-03-20:
Interestingly, importing the STL into tinkercad, making an alteration and then exporting, gave an STL that displayed propelry in CANVAS.

brandonbloch, 2019-03-21:
Hi @Spooner,

Thanks for reporting this! TBH I can't believe it's taken this long for any of us to notice. The issue had to do with incorrect triangle normals in the STL, which would only affect visual preview (importing to Tinkercad and re-exporting probably caused the normals to regenerate). The issue has been fixed, and the correction is now live! Reloading the page, you should now see them being lit correctly.

Cheers,
Brandon

closed 2019-03-21 by brandonbloch