MosaicManufacturing/canvas-issues #100: Make purge block rounded rectangle

wesc23, 2019-05-19:
The rectangular purge blocks can often lift up at the corner due to material contraction, ruining the print.   Rounding the corners would alleviate this greatly.   An alternative is for the purge block to obey the profile brim settings

brandonbloch, 2019-06-07:
Hello,

This has been requested fairly often (either rounded corners or octagonal towers) and we are evaluating it as an option (some people actually report that towers can be harder to remove, and this would make it worse for them).

In the meantime, there is a setting in the Transition section ("Minimum brim count") that is intended to help with this. Because that setting exists, we don't apply the profile's global brim setting to the tower. Let us know if that helps at all or if you're still experiencing issues with that value raised.

Spooner, 2019-09-02:
Adding a brim definitely helped my towers to stick and is better than increasing the bottom layer density.