MosaicManufacturing/canvas-issues #101: Z height / pause

awr4sc, 2019-06-02:
When previewing the print, would it be possible to add a z-height notification attached to the sliders on the side. Or if possible, a location to enter pauses?

something like this:

[image]

I often embed objects into the prints and need to see the z-height, and pause the print. 

Thanks,

brandonbloch, 2019-06-07:
Hello,

We will be adding this (layer height preview with current layer number and Z value) in the next few weeks. Manual pause heights are not currently planned but we will consider the request as well!

brandonbloch, 2019-07-05:
The layer height sliders in the print preview will now display the layer number and Z height when hovering over the slider. I'm closing this issue as your use case is suitably satisfied, but please feel free to open another issue regarding the request for entering pause heights.

Cheers,
Brandon

closed 2019-07-05 by brandonbloch