MosaicManufacturing/canvas-issues #102: Not correctly slicing a model that has been painted

NZSPY, 2019-06-25:
Just having a look at the new paint feature, so I opened an existing project of a 2 colour print of the yin and yang symbol and decided to use the new paint feature to make the "eyes' different colours. 

I used the fill option to do this

The model looks fine on the editing page see screen shot yin1

[image]

but when I slice it the red eye becomes a square see screen shot yin 2 

[image]

and if i move the slice view thru the model it all is the same all the way until the first layer which is completely wrong see screen shot yin3

[image]

brandonbloch, 2019-06-28:
Hi there,

This is a curious one. Can you provide the STLs for this print so we can investigate what's happening here? The fact that the artifacts are not symmetrical makes me suspect that the models may have some mesh errors, but I'd like to take a closer look so we know where to investigate the issue.

Cheers,
Brandon

NZSPY, 2019-06-28:
Hi Brandon, 

Please find attached the STL files, I created these in tinkercad, as an aid to tunning my pallete 

Also here is the link to the model sliced normally as 2 colour and prints fine

https://share.canvas3d.io/projects/bee7a2d632298b79ded2080d12c4bd84 <https://share.canvas3d.io/projects/bee7a2d632298b79ded2080d12c4bd84>

And the model where I painted the eyes in Canvas 

https://share.canvas3d.io/projects/08fe48abf7c0311868d39be424058fdb <https://share.canvas3d.io/projects/08fe48abf7c0311868d39be424058fdb>

Best of luck, really looking forward to this new feature :-)

Cheers
Simon

brandonbloch, 2019-08-15:
Hi again,

Thank you for providing the files to reproduce this. I'm glad to report that we've identified and fixed this issue, and it's now live! You should find that multi-body models which have been painted should slice consistently. Do note that there are a few other issues on this tracker relating specifically to the quality of painted models, but they are unfortunately unrelated to this one and so have not all been fixed yet. That said, please let us know if you still experience any issues with this!

Cheers,
Brandon

closed 2019-08-15 by brandonbloch